tetrabiodistributed/papra #6: Add drawings and dxfs of gaskets

darraghbr, 2021-02-14:
Need to upload and keep updated drawings of the gaskets as well as dxfs, material should be called out on the drawing.

darraghbr, 2021-02-14:
[D-TET-000117-0 Funnel Gasket Drawing.pdf](https://github.com/tetrabiodistributed/papra/files/5978213/D-TET-000117-0.Funnel.Gasket.Drawing.pdf)
[D-TET-000111-0 Fan Box Gasket Drawing.pdf](https://github.com/tetrabiodistributed/papra/files/5978214/D-TET-000111-0.Fan.Box.Gasket.Drawing.pdf)
[D-TET-000114-0 Fan Gasket Drawing.pdf](https://github.com/tetrabiodistributed/papra/files/5978215/D-TET-000114-0.Fan.Gasket.Drawing.pdf)
[D-TET-000115-0 Filter Gasket Drawing.pdf](https://github.com/tetrabiodistributed/papra/files/5978216/D-TET-000115-0.Filter.Gasket.Drawing.pdf)

@mmroden please review when able. I think we need to clarify the material used, the existing build guide says all 2mm craft foam but some of the gaskets in Fusion are 1mm thick. Should probably come up with a more technical term than craft foam as well?

mmroden, 2021-02-14:
I think the Craft Foam term comes from the product we can get either off of Amazon or from Michael's (a craft store in the US).  It's 2mm in its initial form, and then compressed to 1mm, hence the profile in Fusion.

darraghbr, 2021-02-18:
As a group it has been decided to keep the engineering drawings in their parent repos instead of pushing them to the docs repo.

closed 2021-02-18 by darraghbr