tetrabiodistributed/papra #75: add cc-by-4.0 license for docs

capsulecorplab, 2022-06-22:
The code for the repo is licensed under MIT, but the content for the docs itself should be licensed under creative commons, as MIT is not an option for licensing documentation as per the OSHWA certification application.

capsulecorplab, 2022-11-09:
Will probably update the LICENSE file to be a markdown document listing all the licenses used in the papra project, similar to https://github.com/wntrblm/Castor_and_Pollux/blob/main/LICENSE.md

capsulecorplab, 2022-11-20:
Should follow shortcode pattern from https://github.com/twbs/bootstrap/blob/00aa1a5c6e04629955ee45780b86f6ce8fd48ebf/site/layouts/partials/footer.html#L11 for producing license info per page

Burhan-Q, 2022-11-20:
> Should follow shortcode pattern from https://github.com/twbs/bootstrap/blob/00aa1a5c6e04629955ee45780b86f6ce8fd48ebf/site/layouts/partials/footer.html#L11 for producing license info per page

"Better reference"
https://github.com/303adastra/stellardynamics/blob/582f40f3b6c9264a2028744175792eea008e46b4/docs/layouts/partials/site-footer.html

PR #89 (2022-12-01): Add cc by 4.0 license for docs
closes #75
closed 2022-12-01 by mmroden