tetrabiodistributed/papra #116: edits do not cause page refreshes on mac

mmroden, 2023-07-20:
When I edit in either vscodium or hugo on mac, no changes appear to be picked up by the documentation system.  Changes are picked up if I manually kill and restart the server via serve.sh and refresh the page, which is not expected behavior (to automatically refresh the page to reflect changes)

capsulecorplab, 2023-07-20:
This seems to be a Mac-specific issue, as the server rebuilds the site on Linux mint. I can implement a fix to make it work in the [tetra-workspace-image](https://github.com/tetrabiodistributed/tetra-workspace-image/) (which runs cross-platform), but not sure how to fix it specifically for Mac

capsulecorplab, 2023-07-20:
that or we revert podman to docker and just deal with read-only generated build cache issue